zuevval/braille #1: Написать .ino скрипт для калибровки серво

zuevval, 2019-01-25:
Написать скрипт для калибровки позиций сервоприводов в шеститочечной ячейке. Управление через Serial. Поддерживаемые команды:
- [ ] s[цифра]+[число шагов] - в текущей позиции сдвинуть серво на заданное число шагов (лимит 10 шагов для безопасности)
- [ ] s[цифра]+on - переключение в pos_on
- [ ] s[цифра]+off - переключение в pos_off
- [ ] p - напечатать текущие значения pos_on и pos_off для всех серво

alexeyklopov, 2019-01-29:
Completed.

zuevval, 2019-01-30:
> 
> 
> Completed.

Протестировал поверхностно. Нашёл опечатку в Servo_setInside и поставил свои startPos, Steps ([вот в этом коммите](https://github.com/zuevval/braille/commit/d732f91e34cdf71a433634fb90477fc48d26f667)) <br>
Мне кажется или при изменении s<digit>.on+<digit> не изменяется позиция серво, а только цифра? В любом случае, после этой команды серво должен прийти в новое положение (даже если был в позиции off)

zuevval, 2019-01-30:
> Completed.

Но ты молодец

alexeyklopov, 2019-01-30:
> Мне кажется или при изменении s<digit>.on+<digit> не изменяется позиция серво, а только цифра? В любом случае, после этой команды серво должен прийти в новое положение (даже если был в позиции off)

Действительно, в setInside была ошибка. 
Добавил изменение позиции серво при внесении правок в калибровочные массивы.

zuevval, 2019-01-30:
Всё в порядке. Влил в master

closed 2019-01-30 by zuevval