zuevval/braille #10: Воспроизведение голоса в Python

zuevval, 2019-01-29:
С помощью синтеза речи либо воспроизведения заранее записанного голоса обеспечить воспроизведение фраз из набора:
1. 'lesson_1_1': "Урок 1: буквы А, Б, О, П. Длительность - примерно 15 минут"
2. 'lesson_1_2': "Потрогайте точку на поверхности тренажёра. Одной точкой обозначается буква А."
3. 'lesson_1_3': "Буква Б - две точки слева в верхнем и среднем ряду. Ознакомьтесь с ней".
4. 'lesson_1_4': "Прочитайте про себя слоги. БА... АБ".
5. 'lesson_1_5': "Буква О - три точки: вверху слева, посередине справа и внизу слева. Ознакомьтесь".
6. 'lesson_1_6': "Прочитайте про себя слоги и слова. БО... ОБ... БОБ... БАОБАБ..."
Написать функцию pronounce(string_title) <br>
Назначение: обучающая программа (см. [вики](https://github.com/zuevval/braille/wiki/%D0%9E%D0%B1%D1%83%D1%87%D0%B0%D1%8E%D1%89%D0%B5%D0%B5-%D0%BF%D1%80%D0%B8%D0%BB%D0%BE%D0%B6%D0%B5%D0%BD%D0%B8%D0%B5-%D0%B4%D0%BB%D1%8F-%D1%82%D1%80%D0%B5%D0%BD%D0%B0%D0%B6%D1%91%D1%80%D0%B0-(%D0%BF%D0%BB%D0%B0%D0%BD)))

zuevval, 2019-02-05:
Скоро я переделаю этот issue. Тут не нужно знать языки программирования, поэтому, наверное, передам Ивану.

alexeyklopov, 2019-02-05:
Первые 6 уроков уже есть в python/audio/lesson1. Удобно использовать программу Audacity и движок Milena с сайта https://www.nuance.com/omni-channel-customer-engagement/voice-and-ivr/text-to-speech.html#!

zuevval, 2019-02-06:
сделано

closed 2019-02-06 by zuevval