zuevval/braille #47: Синтез речи

alexeyklopov, 2019-02-08:
Перевести всю систему на синтез речи (отпадет необходимость в след. приложениях записывать шаблоны)

zuevval, 2019-02-11:
Это представляется сложным, если хочется работать офлайн. <br>
[Статья на Хабре об онлайн-синтезе](https://habr.com/ru/sandbox/46676/) <br>
[Ответ со StackOverflow на русском об офлайн-синтезе](https://ru.stackoverflow.com/questions/844927/offline-%D1%81%D0%B8%D0%BD%D1%82%D0%B5%D0%B7-%D1%80%D0%B5%D1%87%D0%B8) <br>
Вообще я спрошу у Нины Константиновны и прочих. Может, им приятно слышать не синтезатор, а запись. Фразы же всё равно надо придумывать, записать их, кажется, не так и сложно.

alexeyklopov, 2019-02-11:
Это может пригодиться для озвучки записей в блокноте

zuevval, 2019-04-12:
Вновь интересно (особенно в связи со всё растущим числом аудиозаписей)

zuevval, 2019-04-15:
Windows: найдено перспективное решение - `pyttsx`+`RHVoice` (это, по всей видимости, можно и на Linux) либо `pyttsx` + `Microsoft Elena` (для Windows 10)

[pyttsx tutorial](https://www.devdungeon.com/content/text-speech-python-pyttsx3#change_voice_language)

[download RHVoice](https://nvda.ru/links/rhvoice-0.7.1-sapi5.php)
[RHVoice - все версии, в т. ч. портативная](https://nvda.ru/sintezator-rhvoice#komplekt-rhvoice-v071-sapi5-razdelnye-komponenty)

zuevval, 2019-04-30:
Реализовано нормально, работает.

**Но при запуске в дебаггере выходят ошибки из запущенных потоков**. Это надо когда-нибудь исправить.

closed 2019-04-30 by zuevval