zuevval/braille #51: Склонировать репозиторий

zuevval, 2019-03-02:
Каждому просьба:
1. git clone https://github.com/zuevval/braille.git
1. Проверить, что всё в порядке (попробовать запустить python-код и убедиться, что до Serial-ошибок нет ошибок; открыть любую из ppt-презентаций, чтобы убедиться, что работает LFS).
1. Написать в комментариях под issue о том, что всё работает
1. Кому-то хорошо бы на всякий случай сделать fork

zuevval, 2019-03-02:
У меня всё в порядке.

mualal, 2019-03-06:
Всё работает. Только не воспроизводит аудио с расширением .wav, но воспроизводит с расширением .WAV. <bf>
С расширением .WAV воспроизводит как в плеере, так и из файла audio.py при вызове функции playSoundByFilename

zuevval, 2019-04-30:
С аудио разобрались - это, видимо, была проблема с Git LFS

closed 2019-04-30 by zuevval