zuevval/braille #59: Ввод символов с помощью кнопок

alexeyklopov, 2019-03-07:
В отсутствие подключения к компьютеру выводить символ Брайля по нажатию комбинации клавиш.

- [ ] Нарисовать принципиальную схему сборки кнопок (пока 6, в дальнейшем прибавим пробел).

- [ ] * Проверка подключения к ПК (выводить в сериал символ при подключении и откл.) - сделать при необходимости.

- [ ] Вывод введенного символа в сериал.

- [ ] Расширить функцию listen_joystick до просто listen, обрабатывающую и нажатия клавиш.

- [ ] То же - в меню приложений, но внутри приложений не надо.

- [ ] Проигрывать аудио с названием буквы (в главном меню).

alexeyklopov, 2019-03-08:
Клавиатура:
[image]

alexeyklopov, 2019-03-08:
Кнопки:
[image]

zuevval, 2019-03-10:
> Клавиатура:

Хорошо, issue практически выполнен. А в реальных дисплеях тоже такое расположение? А то интуитивно хочется, чтобы клавиши шли по порядку слева направо

alexeyklopov, 2019-03-10:
За основу была взята реализация с реальных клавиатур Брайля.
К примеру: http://www.nattiq.com/files/braillepen12.bmp

zuevval, 2019-04-12:
пока остановимся на достигнутом

closed 2019-04-12 by zuevval