zuevval/braille #65: Приложение - блиц-опрос

zuevval, 2019-03-10:
По форме issue похож на "приложение - заметки". Обоим им, наверное, предшествует выполнение issue о чтении в Python букв, введённых с клавиатуры.
- [ ] По [Wiki-страничке](https://github.com/zuevval/braille/wiki/App:blitz) проверить, хорошо ли, с вашей точки зрения, Валера описал задачу. Записать в таблицу (Мишину) в common phrases нужные фразы для воспроизведения, отметив "записано" "нет"
- [ ] Реализовать приложение
- [ ] Тестировать на устройстве (на старом, наверное)

DEDBbiTNPAH, 2019-03-20:
Написан временный скрипт блитца, который необходимо переработать для реальной работы в тренажёре

zuevval, 2019-03-24:
> Написан временный скрипт блитца, который необходимо переработать для реальной работы в тренажёре
1. Commit’ы нужно сопровождать содержательным комментарием. Очен похоже, что этот коммит сделан через сайт, что очень нехорошо
2. Пока что голосовое распознавание берётся из listen.py, его и требуется подключить
3. Приложение надо подключить ко всем приложениям в edu_app.py
4. Не нужно файл называть с суффиксом tmp. Обычно tmp в названии содержат файлы, которые можно удалить после завершения работы программы. И нет ничего более постоянного, чем временное

closed 2019-04-30 by zuevval