zuevval/braille #68: Сделать распознавание аудио

zuevval, 2019-03-24:
Поскольку нейросеть, видимо, получить не удастся, сделать альтернативные варианты распознавания.
- [ ] Ознакомиться с файлом python/listen.py и его историей: `git log python/listen.py`, затем `git checkout <firstSixSymbolsOfCommitHash>` - сначала были попытки написать на Pocketsphinx
- [ ] Попробовать исправить вылет с ошибками (путём гугления и т. п), либо написать свой вариант распознавания через speechRecognition
- [ ] Написать два варианта: через google, как сейчас, и через Pocketsphinx (см. также историю readme: `git log python/readme.md`)
- [ ] Подумать над другими возможными вариантами и написать в комментариях под issue

**Не надо задерживаться на отдельных пунктах Issue**. Нужно **быстрое** и **по возможности оффлайн** решение.

DEDBbiTNPAH, 2019-03-26:
За последние два проделана работа:

1)Была собрана информация и проведена попытка реализовать распознавание речи на основе yandex voice recognition. Возникла проблема с API, правильная работа с http-запросом

2)Была скопирована функция распознавание речи с микрофона с помощью pocketsphinx, создан фонетический словарь русского алфавита, найдена русская языковая модель. На данном этапе распознавание букв не удовлетворительно, основная надежда на написание грамматической модели вместо полной языковой, что может повысить точность и скорость распознавания.

Дальнейшая работа покажет возможность распознавания речи на тренажёре, если не удастся получить удовлетворительные результаты, придётся ограничиться обработкой речи на удалённом сервере (google, yandex), что мне кажется вполне обоснованным ввиду сложности задачи

zuevval, 2019-03-27:
> За последние два проделана работа:
>
> 1)Была собрана информация и проведена попытка реализовать распознавание речи на основе yandex voice recognition. Возникла проблема с API, правильная работа с [...]

Пожалуйста, включите в локальный readme папки python информацию о том, что надо для работы pocketSphinx.
В итоговой обёртке всей работы - функции listenSymbol - пожалуйста, вызовите голосовое распознавание от Яндекс; в случае неуспешного запроса распознавайте через pocketSphinx

closed 2019-04-10 by zuevval