zuevval/braille #75: Сделать асинхронный printLine

zuevval, 2019-03-30:
Предлагаю абстрагироваться от конкретного применения и решить задачу.
Дан код:
```
import threading
import time

def f(k=1):
    print(k)
    time.sleep(5)
    print(k)
    time.sleep(5)
    print(k)

def fAsyncBad(k):
    t = threading.Thread(target=f, args=(k,))
    t.start()

fAsyncBad(1)
time.sleep(0.5)
fAsyncBad(2)
time.sleep(0.5)
fAsyncBad(3)
time.sleep(0.5)
fAsyncBad(4)
```
Сейчас этот код выведет 1 2 3 4 1 2 3 4 1 2 3 4. Нужно как-то добиться того, чтобы выводить 1 2 3 4 4 4, т. е. завершать вызовы f по новому вызову. Можно изменять саму f (временные интервалы должны остаться). После вызова f основной поток не должен виснуть, т. е. нельзя запускать синхронную задержку.

MakarSolomatin, 2019-03-31:
import threading
import time

class f:          
    def __init__(self):
        self._running = True;
    def terminate(self):
        self._running = False;
    def __call__(self, k):
        n = 3
        while self._running and n > 0:
            n = n - 1
            print(k)
            time.sleep(5)
current = f();
  
def fAsyncBad(k):
    global current;

    current.terminate();
    current = f();
    t = threading.Thread(target=current, args=(k,))
    t.start()

fAsyncBad(1)
time.sleep(0.5)
fAsyncBad(2)
time.sleep(0.5)
fAsyncBad(3)
time.sleep(0.5)
fAsyncBad(4)

MakarSolomatin, 2019-03-31:
[test.py.zip](https://github.com/zuevval/braille/files/3026689/test.py.zip)

zuevval, 2019-04-15:
issue выполнен, но задача не совсем решена. Serial не thread-safe, его нельзя делить между потоками. В [текущем состоянии ветки experimental](https://github.com/zuevval/braille/commit/09f35b61964f64ea74b07f80ed894469a78629f5) поставлен костыль: работает синхронный код отправки в Serial строки из нулей и единиц; после этого две секунды Питон спит (если спать меньше 1.7, по той или иной причине Arduino прерывает вывод), передаёт управление в listenJoystick с флагом, означающим, что по приходу `+` (выведена буква) тоже надо передать управление обратно в printLine (а не только по приходу `l` `d` `u` `r` и прочих). Если listenJoystick возвращает не `+`, printLine завершает работу

Это также проблема arduino, так что Макару не обязательно вникать... но инициатива приветствуется.