zuevval/braille #87: Блиц - "боевой" вариант

zuevval, 2019-04-30:
Продолжение  issue #65 , #81 
[Wiki с описанием](https://github.com/zuevval/braille/wiki/App:blitz)
- [ ] пользуясь функциями `text_to_speech` из `python/speech_synthetizer.py` (чтобы заработал синтез, см. [инструкцию в python/readme.md](https://github.com/zuevval/braille/blob/experimental/python/readme.md)), `printLine` из `serial_hex.py`, `listen_serial` из `listen_serial.py`, добиться работы приложения "Блиц-опрос" (навигация джойстиком, озвучка, вывод текста на ячейку Брайля).
- [ ] включить приложение в общий список в `main.py`
- [ ] сделать сравнение с рекордом ("Ваш счёт: ..., рекорд: ...")

Commit'ы лучше делать в `experimental`

<hr>
Это на неделю **до 5 мая**. Следующее задание (последнее, без фиксированного срока выполнения): в Блиц-опросе сделать ввод не только голосом, но и с клавиатуры. Потом напишу подробнее, если нужно.

DEDBbiTNPAH, 2019-05-03:
Сделал версию, работающую на ПК, пока что багует на устройстве

zuevval, 2019-05-06:
Теперь работает хорошо. Я не знаю, как будет со старым тренажёром, но с новыми я попытался исправить некоторые несовпадения в нумерации точек, которые были там

closed 2019-05-06 by zuevval